platformio/platform-nordicnrf52 #169: Serial Port Not Being Created

Montvydas, 2023-02-13:
### Problem:
After uploading a simple sketch, that contains `while(!Serial) yield();` within setup(), the serial port isn't being created. Past this point I cannot print anything to the port or access it anyhow.

### Analysis
Using the latest nordicnrf52 platform (currently 9.4.0) I cannot use a serial port, since it isn't being created. For the sketch to compile I also had to install Adafruit TinyUSB Library@1.18.0, although I also tried older versions without success e.g. 1.10.3. As a workaround I've been using nordicnrf52 platform version 8.1.0, which works fine but only compiles with Adafruit TinyUSB Library version up to 0.10.5. I've been wanting to use the latest Adafruit TinyUSB Library, however then I loose the ability to use Serial comms, which is a nono.

There is a possibility that the problem isn't related to nordicnrf52 (maybe tiny USB, maybe Adafruit Bootloader), however hopefully you could direct me to where I could find the culprit or at least you could test on your end that the Serial communication works for you on nRF52840 boards.

### Machine
I have this issue on multiple machines (haven't tried windows or linux):
- Macbook Air M1, MacOSX Ventura 13.2
- Macbook Pro 13 (2015) i5

### Board
I have this issue on multiple boards, including:
- Particle Xenon (nRF52840) with the latest [BSP 0.7.0](https://github.com/adafruit/Adafruit_nRF52_Bootloader/releases).
- Custom nRF52840 based on bootloader based on [Adafruit_nRF52_Bootloader](https://github.com/adafruit/Adafruit_nRF52_Bootloader) (master branch), using latest [Tiny USB](https://github.com/hathach/tinyusb) (e.g. master branch).

maxgerhardt, 2023-02-14:
Please show the exact `platformio.ini` and code. Having to manually install TinyUSB sounds weird. Especially with caveats for TinyUBS like needing `lib_archive =no` that should be handled automatically in the PlatformIO-provided packages.

Montvydas, 2023-02-15:
Forgot to mention, that I am using the Wire library. If I don't use it, it won't complain about not finding Tiny USB. If I do use Wire, this is the result:

Platformio.ini file:
```
[env:particle_xenon]
platform = nordicnrf52
board = particle_xenon
framework = arduino
```

main.cpp file:
```
#include <Arduino.h>
#include <Wire.h>

void setup() {}

void loop() {}
```

If I compile the script I get:
```
Processing particle_xenon (platform: nordicnrf52; board: particle_xenon; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nordicnrf52/particle_xenon.html
PLATFORM: Nordic nRF52 (9.4.0) > Particle Xenon
HARDWARE: NRF52840 64MHz, 243KB RAM, 796KB Flash
DEBUG: Current (blackmagic) External (blackmagic, cmsis-dap, jlink, stlink)
PACKAGES: 
 - framework-arduinoadafruitnrf52 @ 1.10300.0 (1.3.0) 
 - framework-cmsis @ 2.50700.210515 (5.7.0) 
 - tool-adafruit-nrfutil @ 1.503.0 (5.3) 
 - tool-sreccat @ 1.164.0 (1.64) 
 - toolchain-gccarmnoneeabi @ 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 13 compatible libraries
Scanning dependencies...
Dependency Graph
|-- Wire @ 1.0
Building in release mode
Compiling .pio/build/particle_xenon/src/main.cpp.o
Compiling .pio/build/particle_xenon/lib2f6/Wire/Wire_nRF52.cpp.o
/Users/monte/.platformio/packages/framework-arduinoadafruitnrf52/libraries/Wire/Wire_nRF52.cpp:32:10: fatal error: Adafruit_TinyUSB.h: No such file or directory

**************************************************************************
* Looking for Adafruit_TinyUSB.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Adafruit_TinyUSB.h"
* Web  > https://registry.platformio.org/search?q=header:Adafruit_TinyUSB.h
*
**************************************************************************

 #include <Adafruit_TinyUSB.h> // for Serial
          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio/build/particle_xenon/lib2f6/Wire/Wire_nRF52.cpp.o] Error 1
===================================================================== [FAILED] Took 1.06 seconds =====================================================================
```

maxgerhardt, 2023-02-15:
Interesting situation. I remember that the Arduino core developers sprinked includes to Adafruit TinyUSB in various libraries for some reason. In any case, you can either set

```ini
lib_ldf_mode = deep+
```
in the `platformio.ini` **or** add
```cpp
#include <Adafruit_TinyUSB.h>
```
in `src/main.cpp`.

In both cases you should **remove** any manual installation of Adafruit TinyUSB in e.g. platformio.ini/lib_deps. It then compiles normally. 

Sadly the dependency on Adafruit_TinyUSB isn't detected when including `Wire.h` alone, even if `lib_ldf_mode = chain+`.

Montvydas, 2023-02-15:
Cheers, I found out that including Tiny USB in main was what fixed this issue while experimenting myself this morning, but then the suggestion regarding using `deep+` is really what I was looking for!

closed 2023-02-15 by Montvydas

Montvydas, 2023-02-15:
Sorry, one more thing. If I don't include Adafruit_TinyUSB within main.cpp and use Serial:
```
#include <Arduino.h>

void setup(){
    Serial.begin(115200);
    while(!Serial) yield();
}

void loop(){
    Serial.println("Hello World!");
    delay(1000);
}
```
And platformio.ini:
```
[env:particle_xenon]
platform = nordicnrf52
board = particle_xenon
framework = arduino
lib_ldf_mode = deep+
```
I will still get compile error related to not being able to use TinyUSB method:
```
.pio/build/particle_xenon/src/main.cpp.o: In function `setup':
main.cpp:(.text.setup+0xa): undefined reference to `Adafruit_USBD_CDC::begin(unsigned long)'
main.cpp:(.text.setup+0x16): undefined reference to `Adafruit_USBD_CDC::operator bool()'
main.cpp:(.text.setup+0x20): undefined reference to `Serial'
.pio/build/particle_xenon/src/main.cpp.o: In function `loop':
main.cpp:(.text.loop+0x1c): undefined reference to `Serial'
collect2: error: ld returned 1 exit status
*** [.pio/build/particle_xenon/firmware.elf] Error 1
```
I understand I still need to include TinyUSB within main.cpp then, no other way?

maxgerhardt, 2023-02-15:
Yes, if you need `Serial` you need to include Adafruit_TinyUSB.h. The same error also occurrs in the Arduino IDE without it. See https://github.com/platformio/platform-nordicnrf52/issues/119.