platformio/platform-nordicnrf52 #171: BBC:Microbit_V2 LED Matrix doesn’t work on PlatformIO

BenRabor, 2023-03-01:
The issue is described and discussed here

(https://community.platformio.org/t/microbit-v2-led-matrix-adafruit-microbit-library-demo-doesnt-work-on-platformio/31970/1)

The board is **BBC:Microbit_V2**. The Environment is PlatformIO:
Version: **1.75.1** (user setup)
Commit: 441438abd1ac652551dbe4d408dfcec8a499b8bf
Date: 2023-02-08T21:32:34.589Z
Electron: 19.1.9
Chromium: 102.0.5005.194
Node.js: 16.14.2
V8: 10.2.154.23-electron.0
OS: Windows_NT x64 10.0.19045
Sandboxed: No

**PlatformIO IDE - Version 3.0.0**

**Development PC: Windows 10 Pro**

I am testing a simple program to control LED Matrix of bbc:microbit_V2 by manipulating the pins associated with ROWs and COLs of the LED matrix.

My testing program is shown at the bottom of the post.

**However**, after a successful compilation and downloading the program to microbit, I saw that the all led-s in column-3 (counting from 0) could not be lit, they all stay off.

After some more investigation and discussions in the **PlatformIO** and "**microbit.org**", it turned out that the problem is in the definition of the **NRF52832_XXAA** macro in the "**platformio\packages\framework-arduinonordicnrf5\cores\nRF5\SDK\components\device\nrf.h**" File (in lines# 93-95).

After I commented out a few corresponding lines defining **NRF52832_XXAA** macro in this file to look as follows:

```
#if defined (NRF52)
      #define NRF52_SERIES
//   #ifndef NRF52832_XXAA
//        #define NRF52832_XXAA
//    #endif
#endif

```
everything started working correctly.

**NOTE1:** I verified that the same program without any modifications is working just fine in Arduino IDE 2.0.3.
**NOTE2:** The majority of the analysis has been done by **Martin Williams of the support team in 
Micro:bit Educational Foundation:** 

https://support.microbit.org/support/tickets/60359

```
// This is the test to control the 
// Microbit LED MATRIX individually each LED by  
// handling the LED pins (this is the Microbit version dependent)
//
// The code is prototyped from the Adafruit_Microbit Library
// https://github.com/adafruit/Adafruit_Microbit/blob/master/Adafruit_Microbit.cpp
//
#include <Arduino.h> //  needed for PlatformIO 

#if defined (NRF52)
  #warning "NRF52 DEFINED!"
#endif

#if defined (NRF52_SERIES)
  #warning "NRF52_SERIES DEFINED!"
#endif

#if defined (NRF52832_XXAA)
  #warning "NRF52832_XXAA DEFINED!"
#endif

//#elif defined(NRF52) // NRF52833_XXAA)
#define MATRIX_ROWS 5 //!< Number of rows on the microbit LED matrix
#define MATRIX_COLS 5 //!< Number of columns on the microbit LED matrix

uint8_t rowpins[MATRIX_ROWS] = //!< Pin numbers for the rows on the microbit LED matrix
{
    21, 22, 23, 24, 25   // Original from Adafruit Library
}; 

uint8_t colpins[MATRIX_COLS] = //!< Pin numbers for the columns on the microbit LED matrix
{ 
    4, 7, 3, 6, 10 // Original from Adafruit Library 
}; 

uint8_t pixel_to_row[25] =    //!< Defines what row each pixel is in
{
    1, 1, 1, 1, 1, 2, 2, 2, 2, 2, 3, 3, 3,
    3, 3, 4, 4, 4, 4, 4, 5, 5, 5, 5, 5
}; 

uint8_t pixel_to_col[25] =    //!< Defines what column each pixel is in
{
    1, 2, 3, 4, 5, 1, 2, 3, 4, 5, 1, 2, 3, 4, 5, 1, 2, 3, 4, 5, 1, 2, 3, 4, 5,
}; 
//#endif

#define DELAY_VAL 1000

uint8_t matrix_buffer[MATRIX_ROWS] [MATRIX_COLS] = {0};

/// @brief 
/// @param  
void ledMatrixInit(void)
{
    uint8_t c, r;

    for (c = 0; c < MATRIX_COLS; c++) 
    {
        pinMode(colpins[c], OUTPUT);
        digitalWrite(colpins[c], HIGH);
    }

    for (r = 0; r < MATRIX_ROWS; r++) 
    {
        pinMode(rowpins[r], OUTPUT);
        digitalWrite(rowpins[r], LOW);
    }

    for (r = 0; r < MATRIX_ROWS; r++) 
    {
        for (c = 0; c < MATRIX_COLS; c++) 
        {
            matrix_buffer[r][c] = (uint8_t) 0;
        }
    }
} // end-of-ledMatrixInit()

/// @brief 
/// @param  
void clearLedMatrix(void)
{
    uint8_t c, r;

    for (c = 0; c < MATRIX_COLS; c++) 
    {
        digitalWrite(colpins[c], HIGH);
    }

    for (r = 0; r < MATRIX_ROWS; r++) 
    {
        digitalWrite(rowpins[r], LOW);
    }

} // end-of-clearScreen()
 

/// @brief 
/// @param  
 void setPixel(uint8_t row, uint8_t col, uint8_t bright)
 {
    uint8_t b = LOW;

    if (bright == LOW)
    {
        b = HIGH;
    }

    digitalWrite(colpins[col], b); // Change column states for new row
    digitalWrite(rowpins[row], bright);
 } //end-of-setPixel
 
 
/// @brief 
/// @param  
void setup() 
{  
//    uint8_t c, r;

    Serial.begin(115200); 
    Serial.println("\n\r*** Microbit is ready! ***\n\r");

    ledMatrixInit();
} // end-of-setup()

/// @brief 
/// @param  
void loop()
{
    clearLedMatrix();    
    setPixel(0,0,HIGH);
    delay( DELAY_VAL);

    setPixel(1,1,HIGH);
    delay( DELAY_VAL);
	
    setPixel(2,2,HIGH);
	  delay( DELAY_VAL);

    setPixel(3,3,HIGH);
    delay( DELAY_VAL);
    
    setPixel(4,4,HIGH);
    delay( DELAY_VAL);

} // end-of-loop()

```

maxgerhardt, 2023-09-23:
PUSH! This is still a problem!

aiunderstand, 2023-09-23:
Thanks @BenRabor for opening the issue and @maxgerhardt for mentioning it. I have added a small analysis of the problem in the blog mentioned above. 

Bottomline is a line 62 in [nrf5.py](https://github.com/platformio/builder-framework-arduino-nrf5/blob/8521ae7e008ef66bebfc9e813630f89e3fb894a0/nrf5.py) which parses line 10 of the [board definition](https://github.com/platformio/platform-nordicnrf52/blob/c6ae38751c34c7c87521ef8377f107eed424b421/boards/bbcmicrobit_v2.json#L8-L8). This leads to a conflicting pragma definition that overrides the NRF52833_XXAA definition with an incorrect NRF52832_XXAA (see line 93 of nrf.h) which is a completely different uC. This prevents the Microbit V2 to use the second port peripheral, thus limiting the amount of pins and other issues due to it being another chip.

aiunderstand, 2023-09-23:
**For users:** 
Remove the line “mcu” : “nrf52833”, from the board definition file found in eg. C:\Users<username>.platformio\platforms\nordicnrf52\boards\bbcmicrobit_v2.json

**For PR:** 
The quick fix to this problem is to remove line 10 from the board definition. I have not tested this, but it seems that other boards like the nrf52840 are also effected so a better solution would be to address the nrf5.py code. The adafruit boards that use the nrf54840 use a different generator, [adafruit.py](https://github.com/platformio/builder-framework-arduino-nrf5/blob/8521ae7e008ef66bebfc9e813630f89e3fb894a0/adafruit.py) instead of nrf5.py which doesnt have the conflicting pragma.

closed 2023-09-25 by valeros

BenRabor, 2023-09-25:
> completed via 4257ee4

Thank you very much to all of you guys who analyzed and solved this
annoying problem! You are all great! :)

Eugene.

commit 4257ee43 (2023-09-25): Sync Arduino build scripts

This update includes a fix for BBC micro:bit V2 with proper MCU name defines

Resolves #171