platformio/platform-nordicnrf52 #175: Upload issue with 9.6.0 for Arduino Nano 33 BLE

damond2k, 2023-04-28:
Operating System: Windows 10
PlatformIO Version: Core 6.1.6 - Home 3.4.3
Board: Arduino Nano 33 BLE

I noticed that sketches would work fine when uploaded from Arduino IDE, but doesn't when uploaded from PlatformIO. Uploading from PlatformIO will still show as a success, but the device disappears from the COM port immediately after. Regressing to previous releases of nordicnrf52 work fine.

niftydog, 2023-05-02:
Exactly the same here - I've literally lost days on this. Switches to the upload port but never re-establishes the monitor port after upload.

Change platformio.ini to read _platform = nordicnrf52**@9.5.0**_ and suddenly everything works.

jamesldt, 2023-05-31:
I have reproduced the Nano 33 BLE problem on multiple boards (purchased directly from Arduino) in PIO both on Windows 10 and MacOS X High Sierra (10.13.6), both in release and debug build_types, whereas things work perfectly in Arduino on W10 (both with and without a J-Link debugger). 

In tracing the PIO 9.6.0 problem with a J-Link debug probe, it appears the problem lies in the system initialization code. The Nano 33 BLE seems to crash on an stmia instruction with the stack frames containing return addresses back to reset_handler and system_init, which I would say is kind of telling. Also telling, the upgrade between 9.6.0 and 9.5.0 occurred somewhat recently (March 28th), integrating an updated version of the Arduino mbed core (see [here](https://github.com/platformio/platform-nordicnrf52/releases?page=1) ).

Anyway, platform=nordicnrf52@9.6.0 (or, equivalently platform=nordicnrf52) fails as described above, whereas 
platform=nordicnrf52@9.5.0 works fine, both on W10 and MacOS X.  
Wish I'd found this issue report sooner (did the wrong searches way back, drat) - would have saved me weeks of chasing this down. I thought the problem likely came from the pogo pin debug jig I created for myself for connecting with the Nano 33 debug pads. Instead it appears to be a system upgrade problem.

jamesldt, 2023-05-31:
Just to formalize the issue:

**Configuration**
Reproduced on Windows 10 and MacOS X (High Sierra / 10.13.6); 
(probably any host IDE will likely demo the problem)
VSCode (1.78.2)
Python3 (Pre-installed before installing PIO) + Install Certificates.command
PIO (3.1.1, Core 6.1.7, Home 3.4.4)
C/C++ extension (1.15.4)
Nano 33 BLE and Nano 33 BLE Sense (multiple boards, each, purchased direct from Arduino)

**Description of problem**
Upload crashes/hangs on Nano 33 BLE boards when using nordicnrf52@9.6.0 platform.

**Steps to Reproduce**
1      Create new Blink project for Nano33BLE board using Arduino Framework.
2	Build & upload with Nano33BLE960 env below.  Blink fails to start.
3.     Build & upload with Nano33BLE950 env.   Blink correctly starts.

**Actual Results**
Blink program fails to start with env Nano33BLE960, but works with Nano33BLE950.

**Expected Results**
Blink should start on upload.

If problems with PlatformIO Build System:

**Content of platformio.ini**

[env:nano33ble960]
platform = nordicnrf52@9.6.0
board = nano33ble
framework = arduino

[env:nano33ble960]
platform = nordicnrf52@9.5.0
board = nano33ble
framework = arduino

Source file to reproduce issue:
#include <Arduino.h>

void setup() {
  // put your setup code here, to run once:
  pinMode(LED_BUILTIN,OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);
}

void loop() {
    digitalWrite(LED_BUILTIN, HIGH);
    delay(100);
    digitalWrite(LED_BUILTIN, LOW);
    delay(200);
}

renardeinside, 2023-06-27:
I have the same issue with 9.5.0 version too. 
My setup is the following:

1. Hardware 
- Arduino Nano 33 BLE 
- MacBook Pro, 2018
- Arduino connector that comes with the board (USB-A to USB-B Micro)
- USB-A to USB-C converter (so I can plug it into Mac)

2. Software:
```
> pio system info
--------------------------  ---------------------------------------------------------
PlatformIO Core             6.1.7
Python                      3.11.4-final.0
System Type                 darwin_x86_64
Platform                    macOS-13.4.1
File System Encoding        utf-8
Locale Encoding             UTF-8
PlatformIO Core Directory   /Users/renarde/.platformio
PlatformIO Core Executable  /usr/local/bin/platformio
Python Executable           /usr/local/Cellar/platformio/6.1.7/libexec/bin/python3.11
Global Libraries            0
Development Platforms       4
Tools & Toolchains          11
--------------------------  ---------------------------------------------------------
```

3. Project configuration:

```
[env:nano33ble]
platform = nordicnrf52@9.5.0
board = nano33ble
framework = arduino
monitor_speed = 9600

monitor_filters = 
    colorize
    time
```
The code example doesn't play any role - could be a standard blink example.

4. Steps to reproduce:

- Create a new project with the configuration described above
- Plug in the board. The control LED should be green (showing that the connection is in place, but the other LED shouldn't be flashing smoothly (meaning that bootloader mode is not activated).
- Build and upload:

```
 *  Executing task: platformio run --target upload --upload-port /dev/cu.usbmodem14201 

Processing nano33ble (platform: nordicnrf52@9.5.0; board: nano33ble; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nordicnrf52/nano33ble.html
PLATFORM: Nordic nRF52 (9.5.0) > Arduino Nano 33 BLE
HARDWARE: NRF52840 64MHz, 256KB RAM, 960KB Flash
DEBUG: Current (blackmagic) External (blackmagic, cmsis-dap, jlink)
PACKAGES: 
 - framework-arduino-mbed @ 3.5.4 
 - tool-bossac-nordicnrf52 @ 1.10901.201022 (1.9.1) 
 - tool-openocd @ 2.1100.211028 (11.0) 
 - tool-sreccat @ 1.164.0 (1.64) 
 - toolchain-gccarmnoneeabi @ 1.80201.181220 (8.2.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 40 compatible libraries
Scanning dependencies...
Dependency Graph
|-- Blink
|-- IICScanner
Building in release mode
Checking size .pio/build/nano33ble/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  16.6% (used 43608 bytes from 262144 bytes)
Flash: [=         ]   8.3% (used 81844 bytes from 983040 bytes)
Configuring upload protocol...
AVAILABLE: blackmagic, cmsis-dap, jlink, nrfjprog, nrfutil, sam-ba
CURRENT: upload_protocol = sam-ba
Looking for upload port...
Using manually specified: /dev/cu.usbmodem14201
```
The process hangs here. 
Please note that on Arduino IDE it works as expected, therefore I assume the problem is with the pio and not elsewhere.

hkayann, 2023-07-03:
@renardeinside In my case, the issue was `- toolchain-gccarmnoneeabi @ 1.80201.181220 (8.2.1)`. I have replaced the tool with the one that **Arduino IDE** uses. Now, I can upload with any problems.

minayaserrano, 2023-09-24:
Just to follow up and thank you for sharing the workaround. It also happens to me in Ubuntu.
``` 
> pio system info
--------------------------  ---------------------------------------------
PlatformIO Core             6.1.11
Python                      3.8.10-final.0
System Type                 linux_x86_64
Platform                    Linux-5.4.0-163-generic-x86_64-with-glibc2.29
File System Encoding        utf-8
Locale Encoding             UTF-8
PlatformIO Core Directory   /home/minaya/.platformio
PlatformIO Core Executable  /home/minaya/.platformio/penv/bin/platformio
Python Executable           /home/minaya/.platformio/penv/bin/python
Global Libraries            0
Development Platforms       2
Tools & Toolchains          8
--------------------------  ---------------------------------------------
```
Downgrade version to nordicnrf52@9.5.0 fixes the issue, so it is not serious at the moment.
```
[env:nano33ble]
platform = nordicnrf52@9.5.0
board = nano33ble
framework = arduino
test_filter = test_embedded
test_framework = custom
```

jornj, 2023-11-01:
I have verified that the problem is present from tag 9.6.0 to the current 10.2.0

SebastianKumor, 2025-03-28:
This is still a problem for me with latest version of nordicnrf52