platformio/platform-nordicnrf52 #180: Onboard LSM6DS3 not initializing

Xarbenence, 2023-07-24:
Hello,

I have been working with the Xiao Sense nrf52840, but I have unfortunately been dealing with some some issues in getting the onboard LSM6DS3 to initialize when using platform.io. When I flash the "LSM6DS3 High Level Example" script provided by Seeed Studios in the Arduino IDE, the IMU initializes and prints out its data successfully. However, when I flash the same code within Platform.io, using the following platformio.ini file: 
`
[env:xiaoblesense_adafruit_nrf52]
board = xiaoblesense_adafruit
build_src_filter = +<*> -<main_blinky.cpp>
platform = https://github.com/maxgerhardt/platform-nordicnrf52
framework = arduino
lib_deps = 

    https://github.com/miguelbalboa/rfid.git
    https://github.com/Seeed-Studio/Seeed_Arduino_LSM6DS3.git
`

The IMU never successfully initializes, i.e. IMU.begin() doesn't return 0. 

has anyone else had this problem? I'm not sure what's wrong... I know that I'm using the same library, and everything else appears to be functionally sound. 

Like I said, I'm using the dev's example code, but I'll attach it here for reference. Thanks!

`
#include "LSM6DS3.h"
#include "Wire.h"

//Create a instance of class LSM6DS3
LSM6DS3 myIMU(I2C_MODE, 0x6A);    //I2C device address 0x6A

void setup() {
    // put your setup code here, to run once:
    Serial.begin(9600);
    while (!Serial);
    //Call .begin() to configure the IMUs
    if (myIMU.begin() != 0) {
        Serial.println("Device error");
    } else {
        Serial.println("Device OK!");
    }
}

char buf[100];
void loop() {
    //Accelerometer
    sprintf(buf, "Acc Values: x: %f, y: %f, z: %f",myIMU.readFloatAccelX(),myIMU.readFloatAccelY(),myIMU.readFloatAccelZ());
    Serial.println(buf);
    //Gyroscope
    sprintf(buf, "Gyro Values: x: %f, y: %f, z: %f",myIMU.readFloatGyroX(),myIMU.readFloatGyroY(),myIMU.readFloatGyroZ());
    Serial.println(buf);
   

    delay(100);
}

`

gamingdoom, 2025-09-05:
I'm also having the same issue.

gamingdoom, 2025-09-05:
I figured it out, I needed to add
```
build_flags = -DTARGET_SEEED_XIAO_NRF52840_SENSE
```
to my platformio.ini