platformio/platform-nordicnrf52 #183: compilation color output is missing

fariouche, 2023-10-18:
Hello,

I would help to have the color output from ninja. However, the minimum version of ninja that supports color outputs is 1.9.0.
Unfortunately, for linux, the tool-ninja package is 1.7.0 in the registry.

As a test, I modified the builder to add env.AppendENVPath('CLICOLOR_FORCE', '1')
and used my PC ninja executable instead of tool-ninja, and it worked!
Now the following message from the compiler `defined but not used [-Wunused-variable]` with "-Wunused-variable" in nice color.

Can we have an up to date ninja executable available in the registry please? (Mine is 1.11.1)

Thanks

fariouche, 2023-10-20:
exact, I got confused... it's because we use a ninja script to build mcuboot from a custom script... it's not related to nordicnrf52 platform.

valeros, 2023-10-23:
Closing as not related to the to nordicnrf52 platform.

closed 2023-10-23 by valeros