platformio/platform-nordicnrf52 #186: Build/debug freezes at (gdb)

alcohen, 2023-12-17:
When I start a debug session things freeze at (gdb). The last few terminal lines are below.

Listening on TCP/IP port 2331
```
Connecting to target...
Connected to target
Waiting for GDB connection...Reading symbols from C:\Users\amc79.000\source\repos\Tests\GDB Error Debug\.pio\build\adafruit_feather_nrf52840\firmware.elf...done.
(gdb)

```
And there it sits.

Adafruit nrf52840 feather express,

This was working before. I’ve tried it with three projects (including a brand new one), two Feathers, two debuggers, reinstalled platformio and J-link software. Nothing helps.

I did try to on a second PC, and things work properly.

Platformio.ini is:

```
[env:adafruit_feather_nrf52840]
platform = nordicnrf52
board = adafruit_feather_nrf52840
framework = arduino
upload_protocol = jlink
debug_tool = jlink
```
see https://community.platformio.org/t/build-debug-freezes-at-gdb/37295/1 for more info

alcohen, 2023-12-19:
Some more information. I tried uninstalling then reinstalling VS Code. It did not help, although the first time I ran it I think I got the same terminal output but with one or two lines switched, but I didn't take notes. Maybe there's a race condition?

So then I tried a total uninstall of VS Code and config info using [these instructions](https://code.visualstudio.com/docs/setup/uninstall). Again, same outcome (but no switched lines).

valeros, 2023-12-22:
Hi @alcohen, please run the `pio debug --interface=gdb -- -x .pioinit` directly in the IDE terminal and share logs here.

alcohen, 2023-12-22:
I don't know how to access the logs that you're asking for, but here's the terminal output:

```
PS C:\Users\amc79.000\source\repos\Tests\GDB Error Debug> pio debug --interface=gdb -- -x .pioinit
SEGGER J-Link GDB Server V7.70a Command Line Version

JLinkARM.dll V7.70a (DLL compiled Aug 10 2022 
16:32:56)

Command line: -singlerun -if SWD -select USB -device nRF52840_xxAA -port 2331
-----GDB Server start settings-----
GDBInit file:                  none
GDB Server Listening port:     2331
SWO raw output listening port: 2332
Terminal I/O port:             2333
Accept remote connection:      localhost only 
Generate logfile:              off
Verify download:               off
Init regs on start:            off
Silent mode:                   off
Single run mode:               on
Target connection timeout:     0 ms
------J-Link related settings------
J-Link Host interface:         USB
J-Link script:                 none
J-Link settings file:          none
------Target related settings------
Target device:                 nRF52840_xxAA  
Target device parameters:      none
Target interface:              SWD
Target interface speed:        4000kHz        
Target endian:                 little

Connecting to J-Link...
J-Link is connected.
Firmware: J-Link EDU Mini V1 compiled Jun  6 2023 10:50:57
Hardware: V1.00
S/N: 801031983
Feature(s): FlashBP, GDB
Checking target voltage...
Target voltage: 3.26 V
Listening on TCP/IP port 2331
Connecting to target...
Connected to target
Waiting for GDB connection...Reading symbols from C:\Users\amc79.000\source\repos\Tests\GDB 
Error Debug\.pio\build\adafruit_feather_nrf52840\firmware.elf...done.
PlatformIO Unified Debugger -> https://bit.ly/pio-debug
PlatformIO: debug_tool = jlink
PlatformIO: Initializing remote target...     
Connected to 127.0.0.1
Reading all registers
Read 4 bytes @ address 0x00030886 (Data = 0x5100F8D2)
Read 2 bytes @ address 0x00030886 (Data = 0xF8D2)
0x00030886 in ?? ()
Received monitor command: clrbp
Received monitor command: speed auto
SSelect auto target interface speed (2000 kHz)elect auto target interface speed (2000 kHz)  
Received monitor command: reset
RResetting target
esetting target
Received monitor command: halt
Halting target CPU...
...Target halted (PC = 0x00000A80)
Downloading 16032 bytes @ address 0x00026000  
Downloading 2372 bytes @ address 0x00029EA0   
Downloading 8 bytes @ address 0x0002A7E4      
Downloading 476 bytes @ address 0x0002A7EC    
Writing register (PC = 0x   262b8)
Loading section .text, size 0x47e4 lma 0x26000Loading section .ARM.exidx, size 0x8 lma 0x2a7e4
Loading section .data, size 0x1dc lma 0x2a7ec 
Start address 0x262b8, load size 18888        
Transfer rate: 18445 KB/sec, 4722 bytes/write.Read 4 bytes @ address 0x000262B8 (Data = 0x4A0B490A)
Read 2 bytes @ address 0x000262B8 (Data = 0x490A)
Read 2 bytes @ address 0x0002633C (Data = 0xB500)
Temporary breakpoint 1 at 0x2633c: file C:\Users\amc79.000\.platformio\packages\framework-arduinoadafruitnrf52\cores\nRF5\main.cpp, line 79.
PlatformIO: Initialization completed
(gdb) PlatformIO: Resume the execution to `debug_init_break = tbreak main`
PlatformIO: More configuration options -> https://bit.ly/pio-debug
Continuing.
Setting breakpoint @ address 0x0002633C, Size 
= 2, BPHandle = 0x0001
Starting target CPU...
...Breakpoint reached @ address 0x0002633C    
Reading all registers
Removing breakpoint @ address 0x0002633C, Size = 2
Read 4 bytes @ address 0x0002633C (Data = 0xB083B500)

Temporary breakpoint 1, main ()
    at C:\Users\amc79.000\.platformio\packages\framework-arduinoadafruitnrf52\cores\nRF5\main.cpp:79
79      {
(gdb)
```

alcohen, 2023-12-29:
Some more information: I tried this with a different board, a blues.io Swan Feather with ststm32 mcu, and am seeing the same behavior. So it's not nrf52840-related. Is there a way for this issue be moved to another, more appropriate section?