platformio/platform-nordicnrf52 #187: fatal error: mbed_version.h: No such file or directory

mattvenn, 2023-12-18:
I get this error when using the EdgeImpulse arduino libraries. 
They have a system that creates a library which you download and install.

I made a github action and got the same failure:
https://github.com/mattvenn/laura_skirt/actions/runs/7237880190/job/19718051898#step:6:751

`fatal error: mbed_version.h: No such file or directory`

Meanwhile, arduino-cli can compile succesfully, so I believe it's an issue with the platformio side of things.

I was surprised to not find any mention of the issue anywhere so thought I'd file an issue.

valeros, 2023-12-19:
Hi @mattvenn, I didn't find a `platformio.ini` config file in the `firmware` folder https://github.com/mattvenn/laura_skirt/tree/main/firmware. Any hints how to reproduce the issue?

mattvenn, 2023-12-19:
Hi, yes I've removed all the platformio stuff now as I couldn't get it working, check the linked github action, the commit used to run the action has everything you need to reproduce.

commit a095c530 (2023-12-20): Sync Arduino-mbed build script

Resolves #187
closed 2023-12-20 by valeros