platformio/platform-nordicnrf52 #197: Exception handling not working on nRF52840 (RAK4631)

attermann, 2024-05-26:
Throwing any exception causes board to hang.  Supposedly exception handling works with the nRF8240 using the Nordic SDK and with Zephyr, so the chip is capable of it. Not sure what else to try and would really appreciate some help/insight.

Can reproduce with the following minimal code:

```
#include <Arduino.h>
#include <Adafruit_TinyUSB.h>

#include <exception>
#include <stdexcept>

void test(bool dothrow) {
		Serial.println("Test");
		if (dothrow) {
			Serial.println("Pre-throw");
			throw std::runtime_error("runtime_error");
			Serial.println("Post-throw");
		}
		Serial.println("Test finished");
}

void setup() {
  // put your setup code here, to run once:

	Serial.begin(115200);
	while (!Serial) {
		if (millis() > 5000)
			break;
		delay(500);
	}
	Serial.print("Serial initialized\n");

	try {
		Serial.println("Calling test no-throw");
		test(false);
		Serial.println("Called test no-throw");
		Serial.println("Calling test throw");
		test(true);
		Serial.println("Called test throw");
	}
	catch (std::exception e) {
		Serial.println("Caught exception");
	}
	Serial.println("Finished!");

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

Using following `platform.ini`:

```
[env:wiscore_rak4631]
platform = nordicnrf52
board = wiscore_rak4631
framework = arduino
monitor_speed = 115200
upload_speed = 460800
build_flags = -fexceptions
```

Platform/package versions I'm using:

```
CONFIGURATION: https://docs.platformio.org/page/boards/nordicnrf52/wiscore_rak4631.html
PLATFORM: Nordic nRF52 (10.3.0) > WisCore RAK4631 Board
HARDWARE: NRF52840 64MHz, 243KB RAM, 796KB Flash
DEBUG: Current (jlink) External (jlink, stlink)
PACKAGES: 
 - framework-arduinoadafruitnrf52 @ 1.10600.0 (1.6.0) 
 - framework-cmsis @ 2.50700.210515 (5.7.0) 
 - tool-adafruit-nrfutil @ 1.503.0 (5.3) 
 - tool-bossac-nordicnrf52 @ 1.10901.201022 (1.9.1) 
 - tool-openocd @ 3.1200.0 (12.0) 
 - tool-sreccat @ 1.164.0 (1.64) 
 - toolchain-gccarmnoneeabi @ 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 16 compatible libraries
```

valeros, 2024-05-27:
Hi @attermann, there is no such board `wiscore_rak4631` in this repository, are you working with a custom board manifest? Anyway, it doesn't seem like PlatformIO has something to do with the issue, I believe the Adafruit team deliberately disabled exceptions by using the `-fno-exceptions` flag, linking against `newlib-nano` libraries.

attermann, 2024-06-03:
Thanks for the reply @valeros. The board manifest does in fact come from outside of PlatformIO. RAK Wireless provides instructions for patching to add support for their boards.

I was finally able to resolve the issue myself, and you're correct that `newlib-nano` was the culprit. Unlike ESP32, this library built for nRF52 has exception handling disabled, which results in compilations with `-fexceptions` flag succeeding, but apps freezing when an exception is actually thrown.

For anyone finding this issue, the solution for me was to find the platofrm build config script (which on my system was located at `~/.platformio/platforms/nordicnrf52/builder/frameworks/arduino/adafruit.py`) and comment out the line reading "--specs=nano.specs". This causes builds to use the full `newlib` library with exception handling enabled instead of the "nano" version.